ntoll/uflash #4: Make this work with Python 2.

ntoll, 2015-12-21:
What it says on the tin.

funkyHat, 2016-01-04:
It's a pain that it's not possible to specify platform-specific dependencies in requirements.txt

Looks like we're missing at least mock and builtins that could easily be specified as a dependency for pip to fetch, but it would be annoying to clutter up a python3 install with them unnecessarily...

What do you think about creating a separate requirements file for python2?

funkyHat, 2016-01-04:
https://github.com/funkyHat/uflash/tree/python2
Not quite working yet, the hex output is subtly different. The only tests that are failin are test_unhexlify and test_extract, but they are failing *because* the output from hexlify is different.

Perhaps we need more tests to ensure the *output* from hexlify is correct?

(The actual difference is an extraneous `4` on a line on its own at the end of the embedded python script (at least for our current TEST_SCRIPT :) :) ) so it wouldn't actually break anything <_< )

funkyHat, 2016-01-04:
There's a new requirements file just for python 2, can setuptools be configured to grab dependencies from there too if it's on python 2?

funkyHat, 2016-01-13:
I found a nice solution to this: https://caremad.io/2013/07/setup-vs-requirement/
requirements.txt can contain a line like '-e .' and it will look for dependencies inside setup.py. That allows me to check the python version there and just feels neater overall :)

ntoll, 2016-01-13:
Nice spot! ;-)

funkyHat, 2016-01-13:
The problem I have now is Python 2 is resisting my efforts to mock `open` ... (and still the problem where `hexlify` is generating subtly different output)

funkyHat, 2016-01-22:
Ah. Apparently I should have put "fixes #4" instead of "fixes issue #4" in order for github to automatically close this :)

tomviner, 2016-07-26:
This issue can be closed now?

funkyHat, 2016-07-26:
Yep :)

commit 07af284b (2016-08-23): Fix broken 'print' test on Python 2.7. Fixes #4
closed 2016-08-23 by ntoll