ntoll/uflash #7: Add ability to extract Python code from a .hex file.

ntoll, 2015-12-27:
Because sometimes, you don't save the source file... ;-)

funkyHat, 2016-01-04:
So I've written 2 functions for this on https://github.com/funkyHat/uflash/tree/unhexlify

They look good for getting the script back out of a .hex file that was saved somewhere (I'm not sure if it's easy or not to get a flashed .hex back out of a micro:bit so I'm ignoring that for now :)).

The last bit is the command line interface.

I think it makes sense to switch to using argparse (or perhaps click? But argparse is in the standard library). @ntoll what do you think?

ntoll, 2016-01-04:
Send it as a PR and I'll review. THIS IS TOTALLY AWESOME!!!

Agree about arg parsing. Path of least dependencies is good too.

:+1: :smile:

funkyHat, 2016-01-04:
So I've updated my branch with the argparse code and a few more tests. Every time I try to rebase though my "unhexlify()" function is disappearing? Weird.

Still trying...

funkyHat, 2016-01-04:
Ah think I've got it :)

PR #9 (2016-01-12): Implemented extract option to get script out of a hex file
Switched to argparse instead of rolling our own

Closes #7
closed 2016-01-12 by ntoll