ntoll/uflash #10: Getting weird error popup when running on Windows

malemburg, 2016-01-05:
Every time I run the tool, a popup dialog comes up saying something like "There's no disk in the drive. Please insert a disk into driver \Device\Hardware1\DR1" (translated from German). The dialog asks for [Cancel], [Repeat], [Continue]. Clicking on [Continue] let's flashing continue and the tool does write to the correct mounted drive.

malemburg, 2016-01-05:
This likely has to do with the enumeration of drives the script does.

ntoll, 2016-01-05:
@malemburg can you give me details of the version of Python, Windows and the USB set-up you have in that situation? (For example, any other USB devices plugged into the machine.)

I've not used Windows for quite a while but I'll try to recreate the context and see what happens. I may ping Tim Golden to take a look - he's quite brilliant at this sort of thing. ;-)

malemburg, 2016-01-05:
I'm using Python 3.4, Windows 7 and have memory card readers on the machine which show up as drives, but don't have SD cards in them. I presume the enumeration of those are causing the popup.

The Microbit gets mounted as drive F; C and D are hard drives, E, H and I are memory card card slots.

malemburg, 2016-01-05:
Right, that's it. I just tried os.path.exists('E:\\') and get the same popup. I'm not really a Windows programmer, so don't know whether there's a different way to test for readiness of a drive.

ntoll, 2016-01-05:
Apparently there is, but it's all a bit "stand on one leg, stick your finger in your ear then whistle 'God Save the Queen' to make it work" in terms of code comprehensibility. ;-)

(And I'm not sure it'll work either - however, I have enough information now to be able to recreate the problem!)

ntoll, 2016-10-29:
Fixed by #23

closed 2016-10-29 by ntoll