ntoll/uflash #11: Feature: Auto-reset the Microbit after flashing is complete

malemburg, 2016-01-05:
I don't know whether this is possible from the tool, but having the Microbit automatically reset after flashing a new script would be great.

There's a pseudo file on the mounted drive called "HARD_RST.CFG" vs. "AUTO_RST.CFG" which can be used to set this directly on the Microbit, but it only works by going into maintenance mode first (https://www.mbed.com/en/development/hardware/prototyping-production/daplink/daplink-on-kl26z/).

The file mentions sending a "send a serial break command" would have the same effect. Perhaps the tool could do this via the COM port.

ntoll, 2016-01-05:
The device should automatically reset / reboot after being flashed.

Would be interesting to know what you're seeing instead of the reset/restart. :-/

malemburg, 2016-01-05:
I tried going into maintenance mode and deleting the CFG file to toggle the behavior, but this doesn't work as expected. It always comes back in hard reset mode (with the file "HARD_RST.CFG") in Microbit mode.

What I'm seeing is well, nothing :-) The Microbit doesn't do anything and waits for me to press the reset button.

ntoll, 2016-01-05:
Hmm... sounds like you may have either a faulty device or the device has got into a strange state. It is one of the older versions of the hardware, I'll dig out your address and ping you a more up-to-date device.

Can you post me the faulty device..? I'll email you privately with my postal address.

malemburg, 2016-01-05:
Sure, no problem.

GrahamBodley, 2016-10-25:
Is this problem related to the AutoReset problem?

I received my micro:bit this morning, and I appear to have bricked it; I am pretty sure it is a software bug - ie the hardware is fine, and it responds 'normally' in many ways, but it does not behave as the instructions say it should.

I was experimenting with a very simple program - just to display some scrolling text - and it appeared to compile ok, and I was able to download it to the board using Bluetooth; but on completion of the download, the bluetooth handshaking did not complete, and I lost Bliuetooth comms, and no red leds.
No amount of button pokes could get any reaction.

I was able to boot into MAINTENANCE mode, and copy 0234_microbit_if.hex to the device - it did so quite quickly, with the orange LED flickering, then solid on (the instructions say it should flicker continuously, and require an unplug followed by a re-plug without the reset button held down this time).

In case the 'hard reboot' might be required, I removed the file AUTO_RST.CFG to toggle the mode, and sure enough, it got replaced by HARD_RST.CFG : but it still auto-reset itself from MAINTENANCE to MICROBIT, and the device remained bricked.

Deleting the CFG file caused an immediate USB change from MAINTENANCE to MICROBIT, but the mode did not toggle - the CFG stayed HARD_RST. The only way I got it o change was to copy the AUTO_RST.CFG to MAINTENANCE (so both were present), and THEN delete the HARD_RST.CFG.
But it made no difference - the USB still re-enumerated without waiting.

How can I recover from this please?
Graham Bodley

ntoll, 2016-10-29:
@GrahamBodley MicroPython does not support Bluetooth at this moment in time (this is mentioned in the documentation several times). The solution is to flash the device with a non-MicroPython hex from the microbit.co.uk website.

Regarding the device reset - I believe this was a problem with the firmware that has since been resolved.

I'll close this ticket since there's nothing uFlash can do to help with this problem.

closed 2016-10-29 by ntoll