ntoll/uflash #17: requirements so it actually gets found?

ThomasWaldmann, 2016-07-18:
Plugged it in, it's blinking with the demo.
Buttons and sensors seems to work normally (as used in the demo).

Ran:
```
tw@tux:~/w$ virtualenv --python=python3 mipy-env
tw@tux:~/w$ . mipy-env/bin/activate
(mipy-env)tw@tux:~/w$ pip install uflash
Downloading/unpacking uflash
  http://localhost:3141/root/pypi/+simple/uflash/ uses an insecure transport scheme (http). Consider using https if localhost:3141 has it available
  Downloading uflash-1.0.3.tar.gz (273kB): 273kB downloaded
  Running setup.py (path:/home/tw/w/mipy-env/build/uflash/setup.py) egg_info for package uflash
    
Installing collected packages: uflash
  Running setup.py install for uflash
    
    Installing uflash script to /home/tw/w/mipy-env/bin
Successfully installed uflash
(mipy-env)tw@tux:~/w$ uflash
Unable to find micro:bit. Is it plugged in?
```

Ubuntu 16.03, Kernel 4.6.4, Python 3.5.1, USB3 Port (have no USB2).

ThomasWaldmann, 2016-07-18:
It needs to be mounted.

Needed modules: vfat, uas, ...?

ThomasWaldmann, 2016-07-18:
Maybe transform this to a documentation issue "add the required linux modules to docs"?

petrjoachim, 2016-07-18:
I've found the same issue (uflash saying that the device is not found) on Mac OS (10.11.5).

ntoll, 2016-10-29:
Apologies for the delay!

I see what you mean. I'll add something to the docs right now. Thank you for bringing this to my attention: I use Linux but by default my kernel automagically mounts these sorts of things for me. ;-)

See: befa78383edd0a630613753488e22452cc43c826

closed 2016-10-29 by ntoll

ThomasWaldmann, 2016-10-30:
How about putting the kernel modules names into the docs. Like it is now, it is rather vague.

The uas module is rather exotic btw, never seen that before.