ntoll/uflash #26: Add command line flag to watch the script for changes and re-flash

Nagasaki45, 2016-10-28:
I'm currently doing it with `inotifywait`, but I think that it might be usefull as part of uflash itself, with additional command line flag.

```bash
while inotifywait -e modify my_script.py; do
    uflash my_script.py
done
```

(BTW, :+1: for the guidelines for contribution!)

ntoll, 2016-10-28:
Heh... that's really cool. That shouldn't be a difficult change. Do you know enough Python to make this happen..? Happy to review / help etc...

Nagasaki45, 2016-10-28:
Why not. Let me try...

Nagasaki45, 2016-10-28:
Hi @ntoll, some questions:

1. I was searching for ways to do this, and it seems that watching the source file without polling is complicated when the solution must be cross platform ([See this question](http://stackoverflow.com/q/5738442/1224456) for more info). Do you see a reason not to poll the file for changes?

1. Do you want to allow watching for changes when extracting from `.hex`? Will it be usefull in some workflow?

commit 9872792d (2016-10-29): Add --watch flag. Fix #26
closed 2016-10-29 by ntoll