ntoll/uflash #34: Update to latest firmware

dpgeorge, 2017-01-10:
It looks like the firmware of uflash is built from 24th Oct 2016 (correct me if I'm wrong!).  There have been some important bug fixes since then (notably PWM) and it would be great to update the firmware to the latest (also in mu).

carlosperate, 2017-01-10:
As an additional note, since we still don't have proper versioning in the micro:bit micropython, it would be great if at least the CHANGES.rst could contain the hash and date of the micropython version.

ntoll, 2017-01-10:
Hi... I believe the following issue / PR needs addressing before we can update the firmware. The pin update breaks several bot / hardware kits that use the affected pins:

https://github.com/bbcmicrobit/micropython/pull/397

@dpgeorge any thoughts on this..?

dpgeorge, 2017-01-11:
> The pin update breaks several bot / hardware kits that use the affected pins:

Agreed that needs fixing.  See also https://github.com/bbcmicrobit/micropython/pull/398 for a similar issue.

> As an additional note, since we still don't have proper versioning in the micro:bit micropython, it would be great if at least the CHANGES.rst could contain the hash and date of the micropython version.

Yes we definitely need to properly track the exact version of the microbit build.

tim1mw, 2017-04-15:
An update to the uflash firmware would be handy for a project I'm working on. The version supplied in uflash (and mu) crashes if PWM output and the radio receiver are user together. See:

https://github.com/bbcmicrobit/micropython/issues/380

tim1mw, 2017-04-16:
I've created a fork which includes the latest microbit micropython code, see https://github.com/tim1mw/uflash . This seems to solve the PWM+radio crash, but it does exhibit the pin mode problem. I've worked around this in my own code, but I'm assuming this needs to be solved before it can go into the main version of uflash and mu.

ntoll, 2018-03-12:
OK... latest version of firmware is in master and release 1.1 on PyPI.

closed 2018-03-12 by ntoll