ntoll/uflash #41: Failed to create process error

steve-qi, 2018-01-01:
Hi, I'm not sure if I am trying to use this right, but when I try to run uflash I get a Failed To Create Process error.

Any suggestions how I should fix this?

ntoll, 2018-01-01:
What operating system are you using, can you give an example of what you typed in, what the *complete* output of attempting to run the command was and confirm that you're using the latest version of uflash? Thanks.

steve-qi, 2018-01-01:
Hi thanks for coming back on this.  I'm on Windows 10, the pip import seemed to run fine (but only as Administrator).  The install was done today, so I'm assuming it picked up the latest version.  The folder name in the site-packages folder is uflash-1.0.8-py3.5.egg-info.  The date on the uflash.py file is 14/12/2016 though?  I'm trying to run uflash -h in a command prompt (is this the correct way to use it?) and get the message.  I can import the module into a .py file without it complaining.

The full output is:
Microsoft Windows [Version 10.0.16299.125]
(c) 2017 Microsoft Corporation. All rights reserved.

C:\WINDOWS\system32>uflash -h
failed to create process.

ntoll, 2018-01-01:
Hi, I suspect this isn't a bug with uFlash -- rather, you've hit this problem:

https://github.com/pypa/pip/issues/2783

I guess you should upgrade your `setuptools` and try again or upgrade Python perhaps..?

steve-qi, 2018-01-01:
After uninstalling Python and loading it all back on we are in business.  Thank you so much for the pointer.