ntoll/uflash #46: Upstream location of firmware.hex ?

WayneKeenan, 2018-04-19:
Hi,

Please can you tell me where the baked-in default `firmware.hex` is from?   (e.g. which branch or tag in the BBCMicroBit MicroPython repo)

Thanks
Wayne

ntoll, 2018-04-20:
You can use the `os` module to get the Git commit hash relating to the build, like this:

```
>>> os.uname()
(sysname='microbit', nodename='microbit', release='1.0', version="micro:bit v1.0-b'ee6ddc8' on 2018-3-19; MicroPython v1.9.2-34-gd64154c73 on 2017-09-01", machine='micro:bit with nRF51822')
```

The important bit is `version`. The `micro:bit v1.0-b'ee6ddc8'` gives us the commit hash `ee6ddc8`. We can also see that this version of MicroPython for the micro:bit is based on upstream MicroPython at commit hash `gd64154c73`. 

Hope this helps.

WayneKeenan, 2018-04-20:
thanks again :)

WayneKeenan, 2018-04-26:
With what compiler and on what platform is it built?

Sorry to be a pedant.
But for example, if the toolchain uses a more recent newlib then there are ~300 bytes of RAM not available.

Which may be important, and there are perhaps other toolchain imposed differences.

ntoll, 2018-04-26:
This isn't the place to discuss such things (wrong project)... ;-)

You should be asking here: https://github.com/bbcmicrobit/micropython

(Hope this helps).

closed 2018-04-26 by ntoll

WayneKeenan, 2018-04-26:
Sorry if I missed a binary release

carlosperate, 2018-04-26:
To be fair, since there are no binary releases, the origin and environment that built the MicroPython included in uFlash is a valid question for this repository.

ntoll, 2018-04-26:
@WayneKeenan @carlosperate it was built by @dpgeorge -- although when I have the right tool chain installed I just compile them myself as per the instructions in the README at the MicroPython for micro:bit repository.

WayneKeenan, 2018-04-26:
I do compile myself too happily, but my resulting build isn't necesariliy the same thing as what is in the `uflash` repo, I'm not looking for exact, but I'd like to know how my toolchain affects the differences.

I ask not from the point of view of an end-user, per se, but from the point of view of a system integrator building a custom MicroPython hex and trying to understand the implications for an end user  (e.g. available RAM)

Hence my question,  how /is/ the uflash one built?  

What is:
your os? 
your gcc version?
your yotta version?
less so... your cmake version?
others?

Because as I've found (in the MicroPython source) things like the version of newlib actually used effects end user RAM availability.

ntoll, 2018-04-26:
Aha... OK so this is definitely a question for Damien via the repos linked to above. uFlash just flashes micro:bits and has no opinions about the hex file it comes with, IYSWIM.

WayneKeenan, 2018-04-26:
but you ship a default hex with Mu, so that what I'm trying to understand.

WayneKeenan, 2018-04-26:
hence the title :)

ntoll, 2018-04-26:
AFAICT you're asking how the hex was built from MicroPython source in a repos related to a downstream project. My advice is simply to ask back upstream in the MicroPython repos since the person who knows the answers maintains that repository.

ntoll, 2018-04-26:
I want to be very clear... I'm not trying to be unfriendly or fob you off, I just think you'll have a better chance of getting the information you seek if you ask in the MicroPython repos linked to above. I sincerely hope this helps. :-)

WayneKeenan, 2018-04-26:
I'm sorry, I'm not trying to be a troll or pain, I just would like to know for end user reasons, you'll see why soon :)

ntoll, 2018-04-26:
Good for you! And you're certainly not a troll or a pain. I look forward to the fruits of your labours (assuming you get the answers you seek). Hurrah! :-)

dpgeorge, 2018-05-01:
> Hence my question, how /is/ the uflash one built?
> 
> What is:
> your os?
> your gcc version?
> your yotta version?
> less so... your cmake version?
> others?
> 
> Because as I've found (in the MicroPython source) things like the version of newlib actually used effects end user RAM availability.

There's currently no way to track these pieces of information for any given hex file.  And even with all the above info it still might not be possible to make an equivalent build.

I usually use Arch Linux with the latest arm-none-eabi-gcc version.  It was probably gcc 7.3.0.  I think it was the old newlib version so it would have that extra bit of RAM.

While the compiler can subtly change end-user behaviour (eg change in stack usage and available stack for the user) the main thing to be aware of is that newer newlib's will cost 300 bytes RAM.