ntoll/uflash #48: Is there a reason that uflash doesn't preserve filenames?

ScottDWebster, 2018-06-07:
Is there a reason that 'uflash my_file.py' creates micropython.hex instead of my_file.hex?  Would there be an objection to changing it or would it be preferable to add an option like '-k' and/or '--keepname'?  
I would like the functionality to be able to pre-hexlify files (uflash my_file,py ./).  I'll start looking at the code to see how this could be done.

ntoll, 2018-06-07:
Knock yourself out..! :-) The `-k` `--keepname` sounds like a good way to go. Please remember the checklist at the end of here: https://github.com/ntoll/uflash/blob/master/CONTRIBUTING.rst

(BTW.... `micropython.hex` is used simply because uflash usually saves directly onto the device rather than the local FS [nobody has ever considered anyone would want to pre-create/save hex files with uflash]).

ntoll, 2018-06-16:
Great stuff... thank you! Looking forward to getting your PR. :-)

ScottDWebster, 2018-06-18:
I'll have a second pull request soon.  I wanted to have multi-input support (*.py, etc.) and finally decided that the simplest way to implement that was to create a separate script (hexify.py) that loads uflash.py as a module.  I've got it working but I'll need to add tests and documentation.