ntoll/uflash #53: Package for Debian?

petterreinholdtsen, 2018-08-17:
Hi.  I consider packaging uflash and uploading it into Debian.  This might be easier with your help.  Are you interested in helping out to make this happen?

petterreinholdtsen, 2018-08-20:
The request for a Debian package is tracked in https://bugs.debian.org/906450 .

knowledgejunkie, 2018-08-30:
The presence of the included firmware file is triggering some thoughts regarding how uflash/Mu can be packaged for Debian, in terms of

- where the packages can sit in the Debian archive (main vs non-free)
- how much modification the source might need in order to be included in main
- how visible the packages would be if uploaded to non-free
- whether it is possible to remove the requirement to include the firmware (see #54)
- whether we could add uflash as a dependency (Suggests) if uflash were packaged in non-free and a firmware-less Mu was packaged in main

petterreinholdtsen, 2018-08-30:
Is the source of the firmware blob available as free software, and can it be built using only free software?  If so, perhaps it is best to make a separeate package of its source and make a new binary package with the firmware hex file.  It can then be used as a dependency of both uflash and mu, and all three can go into main.

knowledgejunkie, 2018-08-30:
@petterreinholdtsen please track #54 which will hopefully answer questions arising about the MicroPython runtime

knowledgejunkie, 2018-12-21:
Update: earlier this week I pushed my packaging of uflash to the Debian Python Modules repository for review.

ntoll, 2018-12-22:
@knowledgejunkie stonking stuff. Thank you so much for your efforts with this. :+1:

petterreinholdtsen, 2018-12-22:
[Nicholas Tollervey]
> @knowledgejunkie stonking stuff. Thank you so much for your efforts
> with this. :+1:

Hear, hear!  This was a great Christmas present. :)  I eagerly await the
day where we can program Micro:Bit with the tools included in Debian,
also on a deserted island. :)

-- 
Happy hacking
Petter Reinholdtsen

knowledgejunkie, 2018-12-31:
My packaging of uflash has been reviewed and uploaded to the Debian NEW queue. This package currently provides a helper package that will download the current firmware.hex at installation time.

I'm now finalising my packaging of MicroPython for the micro:bit which will provide a firmware.hex built completely from source.

knowledgejunkie, 2019-01-13:
My python-uflash source package was ACCEPTED into Debian unstable yesterday, so I'm happy for this ticket to be closed. The package will migrate into Debian testing in due course.

My separate packaging of MicroPython for the micro:bit is currently being reviewed.

martinbarlow, 2025-03-25:
@knowledgejunkie thanks for getting version 1.2.4 released as debian package.

microbit has a v2 hardware since a while, which requires 2.0.0 of uflash to work. Are there any plans to do a new release?

knowledgejunkie, 2025-05-17:
@martinbarlow Apologies for the delay in responding.

I have not been the maintainer of mu, uflash, or micro:bit MicroPython (or the many build dependencies for yotta) since orphaning them in August 2024.

There is an unresolved upstream bug report [0] stating that micro:bit micropython v1 fails to build with ARM gcc >= 12. (note: Debian Trixie will have gcc 14).

  [0] https://github.com/bbcmicrobit/micropython/issues/795

With regard to Debian, the firmware-microbit-micropython package (for v1 micro:bits) fails to build and has been doing so since at least last year [1].

  [1] https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1073304

A new maintainer has picked up packaging the mu editor, but MicroPython and uflash are still orphaned and have seen no uploads. As a result of this and firmware-microbit micropython failing to build (and therefore migrate into testing), none of these packages will be in the next Debian release "Trixie" as things stand.

The build process for microbit micropython v2 (the v1/v2 universal hex?) looks much simpler but I am not aware of any work to package this yet in Debian. I hope this can be picked up and made available in the next Debian release cycle.

Sorry for being the bearer of bad news; thank you for your interest in micro:bit and mu/uflash.