ntoll/uflash #54: Possible to pull MicroPython runtime from device, update, and then flash?

knowledgejunkie, 2018-08-30:
Is it possible for uflash to pull a runtime from a micro:bit, validate that it is a MicroPython runtime, and then reflash the device after inserting the desired Python scripts, or is the process write-only?

The reason for the query is that the MicroPython runtime, as built via yotta and distributed with uflash/Mu, is built partly (as far I can see following the yotta build dependency chain) from sources that include non-free code (https://github.com/ARMmbed/ble-nrf51822 for example). Please correct me if this is not the case.

If it were possible to pull the runtime off a device prior to flashing, it would mean that the firmware would not need to be distributed in the downstream packages of operating systems that only distribute Free Software.

The MIT-licensed uflash repository contains firmware.hex, and uflash.py contains a copy of the firmware, and is MIT-licensed per header. Can I check that this is compatible with the more restrictive licensing of some of the upstream source(s)?

(I note the separate issue discussing the provenance of the included firmware.hex file at https://github.com/ntoll/uflash/issues/46).

ntoll, 2018-08-30:
Hi @knowledgejunkie,

First of all, thank you for taking the time to bring this up. It's an angle I hadn't even considered. I'm going to cc @jaustin who is both familiar with this project and also work[s|ed] for ARM in the group responsible for yotta. If he can't clarify the situation, he'll probably know someone at ARM who would be able to advise.

@jaustin, the question is if it's OK for the firmware for MicroPython on the micro:bit (built, in part, using yotta which is, apparently, non-free) to be distributed in an MIT licensed project like uflash. I.e. as @knowledgejunkie says, we want to check such a situation is compatible with the more restrictive licensing of some of the upstream source (yotta).

IANAL, but I'm thankful people like you folks are looking at this.

carlosperate, 2018-08-30:
Hi all,

I can probably clarify a few points.

> Is it possible for uflash to pull a runtime from a micro:bit, validate that it is a MicroPython runtime, and then reflash the device after inserting the desired Python scripts, or is the process write-only?

Unfortunately the micro:bit could have any other type of program in the nRF51, so we can't count on MicroPython being present.

> The reason for the query is that the MicroPython runtime, as built via yotta and distributed with uflash/Mu, is built partly (as far I can see following the yotta build dependency chain) from [...]

The DAL pulls libraries from Mbed (I believe they are Apache licensed, but haven't got through all of them) and the Nordic nrf51 SDK, which comes with the Nordic soft devices.

I think the soft device might be the biggest problem here as it is redistributed as a binary blob without source code. However, the MicroPython build is configured to **not** add the softdevice (it uses the `bbc-microbit-classic-gcc-nosd` target). This is because the softdevice contains the BLE functionality and consumes too much memory to run alongside MicroPython, and so all BLE code in the micro:bit DAL has define guards to be compiled-out from MicroPython builds.

@knowledgejunkie I believe all the source code published in those repos have an open source licensed (Mbed classic definitely is). Was the soft device the only worry?

> @jaustin, the question is if it's OK for the firmware for MicroPython on the micro:bit (built, in part, using yotta which is, apparently, non-free) to be distributed in an MIT licensed project like [...]

Yotta is just a build system, which is Apache licensed. In this case the question is about the source code and blobs retrieved by Yotta to build the DAL and MicroPython.

ntoll, 2018-08-30:
@carlosperate awesome response. Thank you. @knowledgejunkie is this enough to answer your query. AFAICT, all the code use in MicroPython for the micro:bit is liberally licensed (@carlosperate, the BBC told me they prefer Apache over MIT, so it makes sense the DAL is Apache).

knowledgejunkie, 2018-08-30:
@carlosperate @jaustin Thank you both very much for your responses and for making explicit the fact that the yotta build target (and final binary) for the micro:bit does not include any SoftDevice binaries.

A key next question is whether it is possible to build the MicroPython runtime with yotta from completely local sources without requiring:

- an mbed account
- a network connection
- SoftDevice binaries present

Having the full source for the MicroPython build chain available and buildable locally is a key requirement for getting projects that include (or strongly depend on) the firmware into Debian main.

(Please note that yotta itself is not yet packaged in Debian.)

knowledgejunkie, 2018-11-16:
Closing this, as building the firmware locally seems possible once yotta and it's missing dependencies are available in Debian.

Thanks for everyone's input.

closed 2018-11-16 by knowledgejunkie