ntoll/uflash #56: Performance penalty if firmware hex string is dropped?

knowledgejunkie, 2018-11-15:
Whilst packaging uflash and the MicroPython firmware, I've been wondering whether there would be a (significant) performance penalty during flashing if the firmware was always read from the local hex file (and the firmware string dropped from the uflash.py file)?

In selfish terms of downstream packaging of uflash, having only one copy of the firmware - in a separate file - would make managing firmware updates very straightforward. In Debian we'd use the Debian-built version of the firmware. Keeping the duplicate copy as an embedded string in the source code not only doubles the size of the package but also creates extra work "injecting" the locally-built firmware back into the source code after each update.

Is performance known to be okay reading the firmware hex in from the local file? Have I missed something important as to why the string version of the firmware is necessary?

carlosperate, 2018-11-16:
> Have I missed something important as to why the string version of the firmware is necessary?

I can't say with certainty, but I always though that the main reason was portability of the module, as this way is easy to move around and integrate in other projects.

ntoll, 2018-11-16:
What @carlosperate says is exactly the reason. It's a single file which can be moved around.

If there is a firmware update then the version number of uflash is also incremented.

knowledgejunkie, 2018-11-16:
@carlosperate @ntoll Thank you both for the confirmation.

The MicroPython firmware will be packaged for Debian separately* to the uflash module/utility, which means locally patching uflash to use the "default-on-Debian" firmware file when called without reference to another firmware image. The _RUNTIME string will be removed with awk/sed.

* Debian does not carry convenience copies of libraries across packages, hence the need to pick mu/uflash apart into their constituent parts

knowledgejunkie, 2019-01-06:
Close this as the packaging of uflash for Debian is now in the Debian NEW queue, pending FTP Master review.

closed 2019-01-06 by knowledgejunkie