ntoll/uflash #64: Check .hex file burning completed in microbit

GauravBole, 2019-10-14:
how we check .hex file burning completed in microbit? to tun another .hex file 
in my case I am running one .hex file and before completing its burring I run another .hex file so I got a error like microbit not connected check microbit .
so if we able to identify first burning is not completed to burn another hex file then i think it solve a problem

ntoll, 2019-10-14:
Hi @GauravBole, thank you for reporting this idea. It depends entirely on the underlying operating system. Flashing to the micro:bit is, as you know, just a filesystem copy of a file onto the device. On some OS's this is non-blocking (i.e. as soon as Python makes the system call to copy the file onto the device, the function returns and the OS does the FS copy in the background). That's why you're seeing the behaviour you describe (Python asks the OS to copy, then immediately exits since the OS gets on with the FS operation in the background).

If you can find a way to do *blocking* file writes with Python on all operating systems, then I think we have a solution to your idea and I would happily merge in any PR you submit if it comes with unit tests and so on. :+1:

carlosperate, 2022-10-11:
Since commit https://github.com/ntoll/uflash/commit/d1b3d03b0abb8f4275607fe93850fe900c52315d uFlash now blocks until the copy is done, so this can be closed as implemented :)

closed 2022-10-12 by ntoll