ntoll/uflash #66: installing uflash gives an error

rhubarbdog, 2020-05-08:
i've just installed uflash with command `sudo -H pip3 install uflash` although it appears to have successfully installed i get the error output
```
Building wheels for collected packages: uflash
  Running setup.py bdist_wheel for uflash ... error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-ied4gn_o/uflash/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmp2uvtutdcpip-wheel- --python-tag cp36:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for uflash
```

carlosperate, 2020-05-11:
Hi @rhubarbdog,

This sounds like it could be an issue with your python/pip installation.
Having a quick look online it's possible you might need to manually install the `wheel` package as well? https://stackoverflow.com/questions/34819221/why-is-python-setup-py-saying-invalid-command-bdist-wheel-on-travis-ci

rhubarbdog, 2020-07-15:
yes i had to issue command `pip install wheel` . `ulfash` also relies on setuptools for which i had to issue command `pip install setuptools` perhaps the README.md and and associated readthedocs documentation

closed 2020-07-15 by rhubarbdog