ntoll/uflash #67: uflash cannot generate hex for microbit v2.

crouse12, 2020-10-24:
I got a microbit v2. I found that uflash cannot generate .hex for microbit v2.
Do you have any plan to update uflash?

ntoll, 2020-10-25:
Hi @crouse12 - yes, this is all in hand and something I've been looking into, but the "universal" hex containing MicroPython isn't quite there yet.

I'm in touch with the micro:bit foundation and will follow their lead when they say things are ready. I'm cc'ing @carlosperate (of micro:bit) just for visibility.

crouse12, 2020-10-26:
Thanks for your help.

carlosperate, 2020-12-21:
@ntoll I had a look at this during the weekend, and I don't think it's worth adding the universal hex into this project.

The attach method only works on v1, so we would need to implement a way to identify v1 vs v2 (the easiest way is using the pyserial package, but that adds a dependency to this pure-stand-along module). And then at that point the only thing uFlash could do on a V2 is to flash MicroPython on its own (would it throw an error if a user tries to flash a python script into a v2?).

crouse12, 2020-12-23:
> @ntoll I had a look at this during the weekend, and I don't think it's worth adding the universal hex into this project.
>
> The attach method only works on v1, so we would need to implement a way to [...]

I think you don't need to identify microbit v1 and v2. You can add some instructions for microbit v2.

kjcole, 2021-02-26:
Any news? Work-arounds?

ntoll, 2021-02-26:
Current master should work for you now. Next Mu will have these fixes too. Thanks and kudos to @carlosperate for making this happen.

carlosperate, 2021-02-26:
Just a small correction, current master doesn't yet work for V2, PR https://github.com/ntoll/uflash/pull/72 still needs more tests to be updated before it can be merged and I haven't had the time to do that yet.
But if you use the uFlash version from that branch it should work.

ntoll, 2021-02-26:
Yeah... sorry @carlosperate - I still need my morning coffee... ;-)

kjcole, 2021-02-26:
Thanks.

My "git-fu" is not great, but if I understand correctly, I should use https://github.com/carlosperate/uflash/tree/fs-cli if I want to `uflash` the v.2.0 today?

carlosperate, 2021-02-26:
Yes, but as that is still a work-in-progress the universal hex is read from the the upy-uh.hex file, it is not included directly in the uflash.py file, so you'd need to have both files available (cloning the repo and checking out that branch should work perfectly fine though).

If you want everything in a single file ready to be moved somewhere else you could use the copy inside the Mu editor: https://github.com/mu-editor/mu/blob/master/mu/contrib/uflash.py
That file has the universal hex added inside the uflash.py file already.

smith9, 2021-04-15:
> Thanks.
> 
> My "git-fu" is not great, but if I understand correctly, I should use https://github.com/carlosperate/uflash/tree/fs-cli if I want to `uflash` the v.2.0 today?

So what does that mean? (I am a developer, but new to python)

using something like 
python setup.py sdist bdist_wheel
to create a package to install locally? and then install from the .whl file

or something like 
python setup.py develop
to install to a local virtual environment.

Then that would give option of using micropython within intellij to flash files to micro bit v2 with the repl (I guess that uflash is a separate dependency for micropython so you can install any version yourself?
or I can see that you can use uflash to watch for saves to a file and then you could load it without using the repl from within intellij

carlosperate, 2022-10-11:
The uFlash v2.0.0 release includes support for micro:bit V2, so this can be considered done 👍

closed 2022-10-12 by ntoll