ntoll/uflash #68: Error extracting Qtruck-BLE.hex: 'utf-8' codec can't decode byte 0x8a in position 1: invalid start byte

Jerry-zirui, 2020-12-08:
I have got this error on Ubuntu 20.04 LTS and Windows 10 muti-time already when ever time I atemp to run "uflash -e xxx.hex xx.py".
Here is one of the HEX file that I trynna to transform to .py.
[Qtruck-A.zip](https://github.com/ntoll/uflash/files/5658052/Qtruck-A.zip)

ntoll, 2020-12-08:
How was the Qtruck hex created..?

Jerry-zirui, 2020-12-09:
Microsoft MakeCode

ntoll, 2020-12-09:
HI @Jerry-zirui , I'm afraid `uflash` only supports MicroPython, not Microsoft's recent variation on a Pythonic theme.

If you want to use Python on the micro:bit, may I respectfully suggest you take MicroPython for a spin (see: https://microbit.org/get-started/user-guide/python/). MicroPython is a full implementation of Python 3 and used on many more devices than just the micro:bit (for instance, Lego's Spike platform).

Best of luck..!

closed 2020-12-09 by ntoll