ntoll/uflash #80: with --runtime disabled, how to upload a new hex file?

g3grau, 2022-02-18:
I just started with a micro:bit and had issues to install mu, so I was happy to find uflash. 
However, trying to test it with an expansion board results in module issues with "superbit". I found a hex file which probably includes this, but as far as I can see, the current uflash (2.0.0) doesn't support uploading a hex file.
Is there another tool to upload a hex file, or is there an alternative to install the module along with my python file?
Or should I just try to use an older version of uflash (which leaves the question open why -r has been deprecated).

g3grau, 2022-02-18:
Hi @ntoll,
thanks for the quick response! First: the superbit is a servo/motor driver addon for the microbit.
The repo is here:  https://github.com/YahboomTechnology/Superbit-expansion-board
I had a hard time .. but finally success running mu in a Win-VM ... this and the fact that I got one v1.5 and a v2 microbit helped to figure out that the hex file for the v2 in the above repo doesn't work ... the version from their homepage differs - so I could run servos with both boards, using Mu in the Win-VM.

So, answering my question about how to upload the hex file seems to be simple: I just need to copy it to the device. So far, so good. Now I'm trying to get rid of the Win stuff (and I can't get any version of Mu running on Suse or Ubuntu).

Comparing the upload times between Mu and uflash, I assume that uflash always uploads some default firmware hex file, is that a correct assumption? That would also explain that uploading the same program which worked from Mu results in an import error for said module. Copying the hex file with the library after doesn't work .. it probably overrides the program?

I'm a confused how this device works, how the python interpreter and the code are installed (and disappear from the directory listing on the device). Is there a good starting point to dig into that?

g3grau, 2022-02-20:
Just now I realize that you also work on mu, @ntoll ... I filed a crash report over there and just managed to remove the web module related code from the source. Now mu works on Linux as well :-)

Regarding my original question ... it now somehow changed to "how to add multiple modules". I realized that there are several add-ons which come with precompiled hex files. So even if a single one can be just copied to the USB drive portion of the microbit, it still doesn't help to combine several modules.

I found another few bits related to "ufs" or MicroFS at https://microfs.readthedocs.io/en/latest/ which seems to imply that one could also upload python modules. I just tried that with some own code and it works :-)
```
# ufs put mylib.py (which contains this:)
from microbit import *
import speech
def greeting():
    speech.say("Yes, it works")

# ufs put main.py (which contains this)
from microbit import *
from mylib import greeting
greeting()     # this is located in a file, uploaded to the user space via ufs put
display.show(Image.HAPPY)   # this is in the main.py
```
After a reset, the board will talk and smile... I'm not sure if multiple files could be uploaded from mu?
At least it seems to work if I upload my mylib.py module via ufs put and then use mu to upload only main.py
from mu.

At the moment I'm hunting for the library sources for the addon board. It seems that this is only available inside the hex file :-/ Not sure if this can be reverted back to python, I mean the part related to the firmware, not the user script. If reverting back to code is not possible, merging library content from different hex files could be an option...I didn't dig into the hex files yet.

carlosperate, 2022-02-21:
> At the moment I'm hunting for the library sources for the addon board. It seems that this is only available inside the hex file :-/ Not sure if this can be reverted back to python, I mean the part [...]

After flashing the modified MicroPython hex file to the micro:bit, if you open Mu and then open the "Files" panel, do you see any python files inside de micro:bit? (on the left side of the panel)
If there are python files there (I assume other than `main.py`), then the Python libraries are included a simple files in the filesystem. If you cannot see any files in that panel then they were embedded inside MicroPython via other methods.

As you have discovered, one of the best methods to send files to the micro:bit with non-standard MicroPython builds is using microFs. Afternatively you can add a add a "custom hex path" to the Mu settings in micro:bit mode. If a path a hex file is added there, every time you flash the micro:bit it will copy the hex file into the `MICROBIT` drive, and then try to send the Python code via microFs.