ntoll/uflash #83: I can't import log

lopezsolerluis, 2023-03-06:
Hi! :)

My problem is that when I upload the firmware with `uflash`, and try to import the `log` module, I get this error:

```
>>> import log
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'log'
```

However, some modules load with no problem, like `import radio` or `import neopixel`.

Am I doing something wrong? Thank you very much in advance!

ntoll, 2023-03-06:
There is no `log` module in the MicroBit version of MicroPython. :disappointed: 

Rather, you should go "old school" and just `print("some message")` and watch the USB serial connection for the output.

lopezsolerluis, 2023-03-06:
Excuse me, but there is: https://microbit-micropython.readthedocs.io/en/v2-docs/log.html

Am I missing something?

lopezsolerluis, 2023-03-06:
In fact, you can `import log` using micro:bit Python Editor (https://python.microbit.org/v/3)

ntoll, 2023-03-06:
Nope... *I'm* the one missing something here. :smile:  

It appears that they've added a new `log` module. @carlosperate do you have any information on this..?

lopezsolerluis, 2023-03-06:
@ntoll Thank you very much!!! :)

j1fig, 2025-05-08:
hello!

If I got this correctly we'd need to update the firmware so as to support the v2 microbit versions (and APIs, such as log, et al)?

Can you provide some high-level instructions as to how to do this @ntoll @carlosperate ? Such that I could do it myself and contribute a patch back?

I'm teaching some teenagers in a month's time and would be happy to take on a little bit of work to have `uflash` support the newer versions (which the students will be running).

j1fig, 2025-05-08:
I have tried locally with this repo and comparing the SHA256 sums of the `firmware.hex` VS the ones from the official microbit [page](https://microbit.org/get-started/user-guide/firmware/), but they seem to be different for sure - so I'm guessing its not as simple as just replacing the firmware.hex file.

```bash
➜  uflash git:(master) ✗ sha256sum firmware.hex
443e179a8d0ce655c5b9c7e8ed0ea4e89f64a0444e0af330c09f87d7cc74c0e2  firmware.hex
➜  uflash git:(master) ✗ sha256sum 0255_kl27z_microbit_0x8000.hex
195b085a16d6074e18f0454cd35de70b5d76718ed80f492858b5abac58590925  0255_kl27z_microbit_0x8000.hex
➜  uflash git:(master) ✗ sha256sum 0257_nrf52820_microbit_if_crc_c782a5ba90_gcc.hex
db21ffa4d665fbccb05b3184f823506f0e2529d53f1b7556b39cd6fbb9204837  0257_nrf52820_microbit_if_crc_c782a5ba90_gcc.hex
```